JK-Rolling/DuinoCoinI2C_RPI #2: bus choice

panayotovip, 2021-09-29:
how i cant specify device bus ?
i need to use /dev/i2c-0 but it is show me only what i have on /dev/i2c-1

JK-Rolling, 2021-10-01:
I'll look into this. see whether is there possibility of using both simultaneously

panayotovip, 2021-10-01:
I thing may by is better option is to have some parameter to point exact an bus.
In others configurations other devices (raspberries) may have other devices (on theirs i2c) with other purposes.
Using multiple bus may appear or create bugs of normal work of device (unexpected reboots, freezes etc. )

JK-Rolling, 2021-10-01:
good input. That should improve stability as well. I'll give the code some test before releasing, so expect some delay

JK-Rolling, 2021-10-02:
enhancement implemented

panayotovip, 2021-10-02:
great work. thanks.

closed 2021-10-02 by panayotovip