JK-Rolling/DuinoCoinI2C_RPI #6: Ring I2C not mining

sagaardu, 2021-10-30:
Hello,

After configuring everything as you indicate, I cannot get the arduinos to mine, leaving constant rejections

```
root@DietPi:~/DuinoCoinI2C_RPI# python3 ./AVR_Miner_RPI.py

Duino-Coin basic configuration tool
Edit Duino-Coin AVR Miner 2.73/Miner_config.cfg file later if you want to change it.
Don't have an Duino-Coin account yet? Use Wallet to register on server.

Enter your Duino-Coin username: valdragas
Enter your choice of I2C bus (e.g. 1): 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
i2cdetect has found I2C addresses above
Enter your I2C slave address (e.g. 3): 3
Do you want to add another board? (y/N): y
Enter your I2C slave address (e.g. 3): 4
Do you want to add another board? (y/N): y
Enter your I2C slave address (e.g. 3): 5
Do you want to add another board? (y/N): y
Enter your I2C slave address (e.g. 3): 6
Do you want to add another board? (y/N): y
Enter your I2C slave address (e.g. 3): 7
Do you want to add another board? (y/N): N
Do you want to add an identifier (name) to this rig? (y/N) y
Enter desired rig name: Nanos
Set developer donation level (0-5) (recommended: 1), this will not reduce your earnings: 1
Config saved! Launching the miner

  Unofficial Duino-Coin RPI I2C  AVR Miner 2.73 2021
  https://github.com/revoxhere/duino-coin
  AVR board(s) on port(s): 3 4 5 6 7
  Developer donation level: 1
  Algorithm: DUCO-S1A  AVR diff
  Rig identifier: Nanos
  Have a peaceful afternoon, valdragas!

17:24:40  sys0  Thank You for being an awesome donator ❤️ Your donation will help us maintain the server and allow further development
17:24:40  net0  Searching for the fastest node to connect to
17:24:40  sys4  AVR mining thread is starting using DUCO-S1A algorithm (4)
17:24:40  sys5  AVR mining thread is starting using DUCO-S1A algorithm (5)17:24:40  sys7  AVR mining thread is starting using DUCO-S1A algorithm (7)17:24:40  sys6  AVR mining thread is starting using DUCO-S1A algorithm (6)17:24:40  net0  Connected to master Duino-Coin server (v2.7)

17:24:41  net0  MOTD: Welcome to the 637 pool (1)
17:24:41  sys3  AVR mining thread is starting using DUCO-S1A algorithm (3)

17:25:36  avr7   Rejected 0/1 (0%)  01.3s  228 H/s  diff 7  ping 16ms
17:25:41  net7  Error connecting to the server. Retrying in 10s (err handling result: [Errno 32] Broken pipe)
17:25:44  sys7  AVR mining thread is starting using DUCO-S1A algorithm (7)
17:25:52  avr4   Rejected 0/2 (0%)  00.1s  221 H/s  diff 7  ping 15ms17:25:52  avr6   Rejected 0/3 (0%)  01.1s  221 H/s  diff 7  ping 13ms17:25:52  avr3   Rejected 0/4 (0%)  01.2s  224 H/s  diff 7  ping 10ms17:25:52  avr5   Rejected 0/5 (0%)  02.7s  224 H/s  diff 7  ping 11ms

17:25:57  net6  Error connecting to the server. Retrying in 10s (err handling result: [Errno 32] Broken pipe)
17:25:58  net0  POOL_INFO: Welcome to the 637 pool (1)
17:25:58  sys0  Periodic mining report (BETA):
                ‖ During the last 77 seconds
                ‖ You've mined 0 shares (0.0 shares/s)
                ‖ Block found: 0
                ‖ With the hashrate of 223 H/s
                ‖ In this time period, you've solved 17217 hashes
                ‖ Total miner uptime: 2 minutes
                ‖ CRC8 Error Rate: 0.0 E/s
                ‖ I2C Retry Rate: 0.727273 R/s17:25:58  net4  Error connecting to the server. Retrying in 10s (err handling result: [Errno 32] Broken pipe)17:25:58  net5  Error connecting to the server. Retrying in 10s (err handling result: [Errno 32] Broken pipe)
[... 99 lines cut ...]
17:31:17  avr3   Rejected 0/24 (0%)  02.5s  182 H/s  diff 7  ping 5ms
17:31:20  sys6  AVR mining thread is starting using DUCO-S1A algorithm (6)17:31:20  sys4  AVR mining thread is starting using DUCO-S1A algorithm (4)

17:31:22  net0  POOL_INFO: Welcome to the 637 pool (1)
17:31:22  sys0  Periodic mining report (BETA):
                ‖ During the last 80 seconds
                ‖ You've mined 0 shares (0.0 shares/s)
                ‖ Block found: 0
                ‖ With the hashrate of 182 H/s
                ‖ In this time period, you've solved 14577 hashes
                ‖ Total miner uptime: 7 minutes
                ‖ CRC8 Error Rate: 0.0 E/s
                ‖ I2C Retry Rate: 0.6875 R/s
17:31:22  net3  Error connecting to the server. Retrying in 10s (err handling result: [Errno 32] Broken pipe)
17:31:26  net0  Connected to master Duino-Coin server (v2.7)
17:31:26  net0  MOTD: Welcome to the 637 pool (1)
17:31:26  sys3  AVR mining thread is starting using DUCO-S1A algorithm (3)
17:32:19  avr7   Rejected 0/25 (0%)  02.1s  182 H/s  diff 7  ping 6ms17:32:19  avr5   Rejected 0/26 (0%)  03.0s  182 H/s  diff 7  ping 7ms

17:32:25  net7  Error connecting to the server. Retrying in 10s (err handling result: [Errno 32] Broken pipe)
17:32:25  net5  Error connecting to the server. Retrying in 10s (err handling result: [Errno 32] Broken pipe)
17:32:28  sys7  AVR mining thread is starting using DUCO-S1A algorithm (7)
17:32:28  sys5  AVR mining thread is starting using DUCO-S1A algorithm (5)
^C17:32:30  sys0  SIGINT detected - Exiting gracefully. See you soon!
```

sagaardu, 2021-11-08:
any solution??

JK-Rolling, 2021-12-04:
is this still happening in latest version? try set `debug=y` in Settings.cfg

JK-Rolling, 2021-12-17:
no activity. closing.

closed 2021-12-17 by JK-Rolling

zolag63, 2022-01-13:
May I open the topic? I faced the same issue, Blocks found: 0
here is the details from the debug:
[image]

zolag63, 2022-01-24:
Sorry I see I uploaded the wrong scrshot, it was saying blocks found: 0.
but actually now it works very well. maybe that was some issue in the network.