JK-Rolling/DuinoCoinI2C_RPI #17: local variable 'i2c_rdata' referenced before assignment

tbillion, 2022-02-24:
04:23:43.290381 DEBUG: 0a: [Errno 121] Remote I/O error
04:23:43.298230 DEBUG: 0a: Reading result from the board
04:23:43.420071 DEBUG: 0a: [Errno 121] Remote I/O error
04:23:43.429786 DEBUG: 0a: Retrying data read: local variable 'i2c_rdata' referenced before assignment
04:23:43.439259 DEBUG: 0a: Sending job to the board
04:23:43.447743 DEBUG: 0a: Job: 4fdbe284673c1b2bbc86762d225634e20c3fa8f6,160d4ab63735e127bacb14d41eab21424252db7c,6,

04:23:43.550828 DEBUG: 0a: [Errno 121] Remote I/O error

JK-Rolling, 2022-03-18:
the worker is which board? e.g. nano/pico

tbillion, 2022-03-18:
Pico

JK-Rolling, 2022-03-22:
have you try this workaround?
https://github.com/JK-Rolling/DuinoCoinI2C_RPI#arduino-mbed-os-rp2040-version-280

tbillion, 2022-03-22:
Yes . I did that

. I assume that's another problem entirely however that hack doesn't stop
it from hanging the bus.

And it doesn't fix the variable thing . You may have fixed the variable
thing because when I re down loaded last night it was no longer an issue.

The hanging the bus thing on the mbed platform is a very valid issue
however

I have attempted this code on a miner and it will start the first one but
it will not run after that it just infinitely times out

JK-Rolling, 2022-03-23:
maybe try fallback to mbed os 2.3.1 ?

tbillion, 2022-03-25:
Ok bricked one environment trying to fall back to mbed 2.3.1 rolled another environment back successfully and now all is functioning . Sorta.

Is the status quo going to be to have to run it on mbed 2.3.1? 

Is there a way to set the difficulty in the main script so that I don't have a bunch of incorrect difficulty errors. It will run for a few minutes and then it will say incorrect difficulty then as it continues it will go for a while then right back to difficulty incorrect errors , my accuracy is horrible like in the high 30% I only have 10 picos hooked to the back plane and there is room for 32 is this a timing issue?
[image]
[image]

tbillion, 2022-03-25:
[image]

JK-Rolling, 2022-06-28:
the Pico sketch no longer use mbed now. try update the Pico to latest sketch and use the latest Python script.
the update enabled dual core functionality and better error correction

JK-Rolling, 2022-06-30:
no response. closing.

closed 2022-06-30 by JK-Rolling

tbillion, 2022-07-05:
Nah it still didn't work . Not sure what 8t 8s not solved