JK-Rolling/DuinoCoinI2C_RPI #19: Request: ESP or PicoW master

tbillion, 2022-07-30:
is there a way to modify the master code to make an esp8266 or picoW to be the master? Pico W would be preferred because this will just drop into the board i designed . would make it pretty simple for the dual core W master also to control both buses and control them at the same logic level 3v3

JK-Rolling, 2022-08-06:
yes for esp8266. picoW however, no Arduino compatible library available yet except for microPython

DocYester, 2022-08-14:
Would be very interesting to have an esp8266 as i2c master device.
Way less wlan connections.

tbillion, 2022-08-14:
I mean in general that's what ricuans code does makes the esp device the master. I was just wondering if there was a way to use this improved i2c code with the same hardware config .

JK-Rolling, 2022-09-06:
ESP8266 is a pretty bad and unreliable I2C slave (see inserted image below, left is the I2C master, right is the I2C slave). After looking into some other communication protocol and did some sanity test, I seeing better result.

I'd imagine a new repository will be created in the future after proof of concept stage

[image]

closing this case won't fix.

closed 2022-09-06 by JK-Rolling