JK-Rolling/DuinoCoinI2C_RPI #24: negative mining

NoNameNoCry, 2022-09-18:
Hi!
@JK-Rolling how are you doing?
I ran into an interesting problem that after an hour of work, my farm began to dig at a loss)) Can you help me?)
[image]

Balance 3 hours ago 2584.00371871

[image]

NoNameNoCry, 2022-09-25:
[image]
Help me pls

JK-Rolling, 2022-10-07:
please email to duino.coin <at> gmail.com or ask in duino-coin discord about your issue. I believe the owner might be able to help you there. closing as I do not have control over the issue

closed 2022-10-07 by JK-Rolling