JK-Rolling/DuinoCoinI2C_RPI #26: problem with arduino compatibility board

GremlineQ, 2023-03-09:
I use the seeeduino xiao board which is compatible with arduino without removing a few lines it does not compile, it is not an insurmountable obstacle but I report it anyway

JK-Rolling, 2023-04-30:
need more info like the error messages, your proposed solution etc

GremlineQ, 2023-04-30:
I do not know a good solution, I only know that even when it was turned off, it still did not allow me to compile, I had to remove a few lines that were displayed as an error, and the functions that blocked it were the wathdog and the built-in temperature sensor, if I remember correctly. I will point out that both items were disabled in the user configuration, but still gave an error. When I have time, I will write what exactly I had to remove

JK-Rolling, 2023-05-01:
cool! looking forward to your update

JK-Rolling, 2023-06-24:
I'll close this case by end of July if there's no update

closed 2024-06-21 by JK-Rolling