JK-Rolling/DuinoCoinI2C_RPI #27: Newcode release 3.5

Pheal22, 2023-05-08:
I have been mining quite successfully with the RPI Pico mining rig for sometime now and was wondering if there will be any update on the code to reflect the new 3.5 release on the Duino site?  I see that the Pico mining rig software uses v3.3 so there is no issue at the moment but also see that there is an efficiency increase with the new  v3.5.

JK-Rolling, 2023-05-13:
although the hashrate increased, but several workers reported issue where Kolka assigned incorrect difficulty level. 
Anyway, I'll look into incorporating the enhancement soon

Pheal22, 2023-05-13:
No problem, just saw the 3.3 upgrade and using that now – all OK.  Many thanks for all your hard work, really love the I2C rig!

JK-Rolling, 2023-06-24:
I2C master software updated to 3.5. Thank you and closing

closed 2023-06-24 by JK-Rolling